Pioreactor/rpi_hardware_pwm #1: Invalid argument Error on change_frequency

tobiaszehntner, 2021-06-09:
Hi there,

When trying to change frequency I get an Invalid Argument error:

```
p = HardwarePWM(1, hz=200)
p.start(0)
p.change_duty_cycle(80)
p.change_frequency(50)
Traceback (most recent call last):
  File "/home/pi/.pycharm_helpers/pydev/_pydevd_bundle/pydevd_exec2.py", line 3, in Exec
    exec(exp, global_vars, local_vars)
  File "<input>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/rpi_hardware_pwm/__init__.py", line 98, in change_frequency
    self.change_duty_cycle(self._duty_cycle)
  File "/usr/local/lib/python3.7/dist-packages/rpi_hardware_pwm/__init__.py", line 91, in change_duty_cycle
    self.echo(dc, os.path.join(self.pwm_dir, "duty_cycle"))
  File "/usr/local/lib/python3.7/dist-packages/rpi_hardware_pwm/__init__.py", line 66, in echo
    f.write(f"{m}\n")
OSError: [Errno 22] Invalid argument
```

CamDavidsonPilon, 2021-08-17:
Hi wow sorry I missed this! 

Yes, what happened was this error: https://stackoverflow.com/questions/23050738/beaglebone-pwm-limited-to-100-hz/23050835#23050835

However, this library should figure this out for you! (Or aleast provide a better error message).

CamDavidsonPilon, 2021-08-17:
👋 should error be fixed (i.e avoidable) in the latest version 0.1.1

closed 2021-08-17 by CamDavidsonPilon