Pioreactor/rpi_hardware_pwm #5: Frequency Setting off target

DanMan32, 2022-07-17:
Using Rigol DS1054Z oscilloscope to measure the PWM output, I found to get 50.0hz, I had to set HardWarePWM to 46.7hz.
Setting function to 50hz, which is what is expected for SG-90 servos (really 20mS period), I was getting an output of 53.6hz or 18.66mS.

Otherwise I found this really easy to use and much more stable than software PWM that comes with Python library RPi.GPIO.
Too bad RPi only comes with 2 PWM channels, shared with the audio features of the RPi.  I would have loved to use this for controlling RGB LED and get more stable results when using vPython.

CamDavidsonPilon, 2022-07-17:
👍 this is useful. I never tested this with a scope, but now I really would like to track down this bug.

CamDavidsonPilon, 2022-07-19:
Hm, I'm not able to reproduce the issue. I'm using a Saleae digital analyzer. Setting the frequency=50hz and dc=10% on hardware PWM channel 1:

[image]

Similar results with PWM channel 0. 

I'm using an RPi 3A. What are you using?

BasVDriel, 2023-02-13:
Same issue. DS7102V oscilloscope. A frequency of 30.0Hz at 10% duty cycle has an actual frequency of 32.1Hz.
This is on the raspberry pi 4B, using channel 0

CamDavidsonPilon, 2023-02-13:
Huh, okay, so something is definetly up. My partner has a beefy scope, I'll get them to help repro this on our end.

BasVDriel, 2023-02-15:
Quick update with my findings. When I cat /sys/class/pwm/pwmchip0/pwm0 it returns the exact period of 35695236. However, when I  clone the repository from github and try the same settings (30Hz at 10%)  I get the correct value of 33333333. So this only arises when I try to use the pip package.

CamDavidsonPilon, 2023-02-15:
Very useful, thanks @BasVDriel. Running `vcgencmd measure_clock pwm` on an Rpi 3 gives 100000000. So I think the difference is due to Pi 4s vs Pi 3s. 

I'll spend some time this week / next week fixing this.

synman, 2023-02-15:
What a great find!   For those who need precision this sounds like the way to go and it shouldn't be too hard to implement, assuming the overhead of the lookup is minimal.

BasVDriel, 2023-02-15:
Yeah all good, I'd make a pull request if I had time to fix it. But, if  someone has a fix ready I can test it on the scope to test the real world accuracy. I only have a Pi 4 though.

CamDavidsonPilon, 2023-03-05:
Follow-up: working on this issue this week

CamDavidsonPilon, 2023-03-05:
Hm, so I had a moment to test with with an RPi4, and I'm still not able to reproduce it with either hardware pin. Some examples of what I'm seeing:

[image]

[image]

[image]

----

Some more details:

1. `vcgencmd measure_clock pwm` returns `frequency(25)=99997560`
2. `cat /proc/device-tree/model` returns `Raspberry Pi 4 Model B Rev 1.4`
3. This Pi is doing no overclocking. Here's a gist of the `/boot/config.txt`: https://gist.github.com/CamDavidsonPilon/c578170ba6f092f0fd6b0ecfbc97b5a4

I would appreciate it if other users could provide the same details!

BasVDriel, 2023-08-03:
1. frequency(25)=107143064
2. Raspberry Pi 4 Model B Rev 1.5
3. No overclock indeed