Pioreactor/rpi_hardware_pwm #10: dtoverlay=pwm-2chan has no effect on RPi 5

schech1, 2024-01-12:
I added dtoverlay=pwm-2chan to /boot/config.txt on a Raspberry Pi 5, but` lsmod | grep pwm` does not show any output.
Also there is no pwm-signal generated either on GPIO19 or 18.

CamDavidsonPilon, 2024-01-12:
Hi @schech1, are you using the latest version (0.2.1, check with `pip list`). If not, please upgrade. The RPi5 is supported by adding `chip=2`, ex: `HardwarePWM(..., chip=2)`

schech1, 2024-01-12:
Yes, I`m using 0.2.1 and i also used `chip=2`.
I also believe that issue is not really related with rpi_hardware_pwm.
But should I see `pwm_bcm2835` when using `lsmod | grep pwm` on a RPi5 after reboot?

CamDavidsonPilon, 2024-01-12:
Testing on my Rpi5, no, I don't see `lsmod | grep pwm` return anything

schech1, 2024-01-12:
I'm trying to figure out what I do  wrong.
I connected an LED to GPIO 18 and ran this simple test
:
```
from gpiozero import PWMLED
from rpi_hardware_pwm import HardwarePWM
from time import sleep

# Not working
pwm = HardwarePWM(pwm_channel=0, hz=60, chip=2)
pwm.start(100) 
sleep(2)
pwm.stop()

# Working
led = PWMLED(18, frequency = 200) 
led.value = 1
sleep(1)
led.value = 0
```

Using `gpiozero` on the same pin, it works. 
Is there anything I can do to investigate further?

CamDavidsonPilon, 2024-01-12:
hm, try the following in dtoverlay (this is from my RPi5):

```
dtoverlay=pwm-2chan,pin=12,func=4,pin2=13,func2=4
```

then try the same code, but targeting pin 12. Maybe something is different with the other pins?

schech1, 2024-01-12:
Yeah, that works.
Got it working no on pin 12 and 13.

wayneOS1337, 2024-02-06:
I still got problems on Pi5.

config.txt:
dtoverlay=pwm-2chan,pin=12,func=4,pin2=13,func2=4

python:
HardwarePWM(pwm_channel=1, hz=frequency, chip=2)

I get no PWM output on Pin 13. What am i missing? Thanks in advance.

CamDavidsonPilon, 2024-02-06:
@wayneOS1337 not sure - did you do a reboot? Can you paste here your entire config.txt file?

CamDavidsonPilon, 2024-02-06:
There is an errant ` at the end of the last line of your config.txt: is that a type in github or your config?

wayneOS1337, 2024-02-06:
Yes, sorry that was a typo in my comment. i edited my comment. The file is without it.

CamDavidsonPilon, 2024-02-07:
Not sure! How are you measuring PWM on a pin?

wayneOS1337, 2024-02-07:
I tested it with a Servo (same settings as on the Pi4) and also with a LED. Did you use Bookworm as OS? or is there any additional Software i need to install?

valtsu23, 2024-03-04:
I have the same issue with Rpi 5. Even with "dtoverlay=pwm" "lsmod | grep pwm" gives nothing. 
"pinctrl get 18" gives: 18: a3    pd | lo // PIN12/GPIO18 = PWM0_CHAN2

I see this error on startup: 
`clk: couldn't set clk_pwm0 clk rate to 100000000 (-22), current rate: 30720000`

valtsu23, 2024-03-05:
Problem is pwmchip index. 
https://forums.raspberrypi.com/viewtopic.php?t=359251&sid=5c309b2e5afd4af2b928bda747f4296b#p2154764

CamDavidsonPilon, 2024-03-06:
I've released a 0.2.2 version that allows for more chips, thanks to @colinpate for the PR

bbfjumper, 2024-03-13:
Hi there, I`m having a simliar issue. 

the command `pwm = HardwarePWM(pwm_channel=0, hz=100, chip=2)` throws the follwing error: 

`[Errno 22] Invalid argument
File "/home/raspi4/Projects/scripts/control_motor.py", line 7, in set_motor_speed
    pwm = HardwarePWM(pwm_channel=0, hz=100, chip=2)`
[config.txt](https://github.com/Pioreactor/rpi_hardware_pwm/files/14588293/config.txt)

    
See attachement for /boot/firmware/config.txt file

Thanks a lot for helping!

valtsu23, 2024-03-14:
Hi, 
Try this: pwm = HardwarePWM(pwm_channel=2, hz=100, chip=2)

bbfjumper, 2024-03-14:
Hi, thanks for the fast replay.

that leads to another error: 
`Exception has occurred: OSError
[Errno 22] Invalid argument`

my full code is:
```

from rpi_hardware_pwm import HardwarePWM
import time
from loguru import logger

def set_motor_speed(speed: int) -> None:
    pwm = HardwarePWM(pwm_channel=2, hz=100, chip=2)
    pwm.start(100)  # full duty cycle
    time.sleep(3)
    pwm.change_duty_cycle(20)

    pwm.stop()

    duty_cycle = 10 + (speed / 9000) * 80

def stop_motor() -> None:
    pass
    # pwm.change_duty_cycle(10)
    # pwm.stop()

def read_values() -> tuple:
    pass

if __name__ == "__main__":
    set_motor_speed(9000)
```

CamDavidsonPilon, 2024-03-15:
@bbfjumper perhaps is a kernel issue: https://forums.raspberrypi.com/viewtopic.php?t=367294 (a new kernel was released in the 24-03-12 image)

I'll be watching that thread for more

bbfjumper, 2024-03-15:
okay that could be the solution, i assume downgrading would solve the problem, what kernel version are you using?

CamDavidsonPilon, 2024-03-15:
I'm using the 6.1.63, from the 2023-12-11 release.

l4cer, 2024-03-16:
I'm having the same invalid argument issue with my Raspberry Pi 5 kernel 6.6. Could you instruct me on how to downgrade the kernel to 6.1.63? When I use the legacy version of the 6.1 kernel from Raspberry Pi Imager v1.8.5 for RPi 5, I just can't finish the boot...

CamDavidsonPilon, 2024-03-16:
> When I use the legacy version of the 6.1 kernel from Raspberry Pi Imager v1.8.5 for RPi 5, I just can't finish the boot...

Try downloading an older image from here: https://www.raspberrypi.com/software/operating-systems/

[image]

bbfjumper, 2024-03-21:
with downgrading my kernel to 6.1.63 it works, thanks everbody for helping me. 

I downgraded with `sudo rpi-update 3cfd50755ffbb2f952408b59df1a9a63d3af5387`
with SHA from: https://github.com/raspberrypi/rpi-firmware/commits/master/?after=f63715223f7cf0ce0cb3be57361ecab8d917699c+34

valtsu23, 2024-03-21:
The Kernel issue seems to be fixed in 6.6.22 version. I had the same error with 6.6.20, but not with 6.6.22.

geekworm-com, 2024-03-23:
Yes, after updating the kernel from `6.6.20` to `6.6.22`, the PWM fan started working properly.

Schwarzkopf-Henkal, 2024-07-14:
Encountered same Invalid Argument issue with my Raspberry Pi5 on Ubuntu 24.04 Kernel 6.8.0.
```
Linux ubuntu 6.8.0-1007-raspi #7-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun 24 10:21:12 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```
My code is as followed
```python
from rpi_hardware_pwm import HardwarePWM
from time import sleep
pwm=HardwarePWM(0,60,chip=2)#channel 0 1 2 3 for GPIO12 13 18 19 respectively
pwm.start(100)
pwm.change_duty_cycle(0.5/10)
# pwm.change_frequency(25_000)
sleep(2)
pwm.stop()
```
added `dtoverlay=pwm-2chan,pin=12,func=4,pin2=13,func2=4` to config.txt

When I execute the code without sudo, I got error log as followed
```
Traceback (most recent call last):
  File "/home/pi/C2/hardwarepwm.py", line 3, in <module>
    pwm=HardwarePWM(0,60,chip=2)#channel 0 1 2 3 for GPIO12 13 18 19 respectively
        ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/rpi_hardware_pwm/__init__.py", line 56, in __init__
    raise HardwarePWMException(f"Need write access to files in '{self.chippath}'")
rpi_hardware_pwm.HardwarePWMException: Need write access to files in '/sys/class/pwm/pwmchip2'
```

When I execute the code **with** sudo, I got error log showing Invalid Argument
```
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/C2/hardwarepwm.py", line 5, in <module>
    pwm.change_duty_cycle(0.5/10)
  File "/usr/local/lib/python3.12/dist-packages/rpi_hardware_pwm/__init__.py", line 105, in change_duty_cycle
    self.echo(dc, os.path.join(self.pwm_dir, "duty_cycle"))
  File "/usr/local/lib/python3.12/dist-packages/rpi_hardware_pwm/__init__.py", line 78, in echo
    with open(file, "w") as f:
OSError: [Errno 22] Invalid argument
```

alectutin-qut, 2024-09-13:
For anyone experiencing this issue, following the instructions on [this Gist](https://gist.github.com/Gadgetoid/b92ad3db06ff8c264eef2abf0e09d569) worked for me!