Pioreactor/rpi_hardware_pwm #17: Cant get to install rpi-hardware-pwm

electro201, 2024-09-19:
Good Day all,

I want to activate and try out hardware pwm on my raspberry pi 5. I followed the instructions given here

https://github.com/Pioreactor/rpi_hardware_pwm

and made the necessary modification to /boot/firmware/config.txt and saved and rebooted. When I used

lsmod | grep pwm

It was showing 

 

> pwm_fan                         49152    0

I couldnt get pwm_bcm2835.

Then I tried to install rpi-hardware-pwm using

`sudo pip3 install rpi-hardware-pwm`

It gives this error:

> error: externally-managed-environment
> 
> * This environment is externally managed To install Python packages system-wide, try apt install python3-xyz, where xyz is the package you are trying to install.
> 
> If you wish to install a non-Debian-packaged Python package, create a virtual environment using python3 -m venv path/to/venv. Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make sure you have python3-full installed.
> 
> For more information visit http://rptl.io/venv
> 
> note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking you Python installation or OS, by passing --break-system-packages. hint: See PEP 668 for the detailed specification.

Could someone please tell me what I am doing wrong here. How to install this rpi-hardware-pwm in my Raspberry Pi 5. It is running Raspberry OS with all the updates installed.

Thank you.

CamDavidsonPilon, 2024-09-20:
Try 
```
sudo pip3 install rpi-hardware-pwm --break-system-packages

```

to install this library.

electro201, 2024-09-20:
CamDavidsonPilon,

Many thanks for your response. Actually I found out what I was doing wrong in this thread.

https://forums.raspberrypi.com/viewtopic.php?t=376868

I made a virtual environment using venv and then tried to install inside it, then it works. 

I got it working now. Thanks again for all your great work on this.

closed 2024-09-20 by electro201