Pioreactor/rpi_hardware_pwm #22: config.txt moved from /boot to /boot/firmware

electro201, 2025-04-05:
Good Day,

I desperately need some help here and I believe I can sort this thing out.

I am using a Raspberry pi 5 and running the latest Rasbian/Bookworm OS. I used it to run PWM programs using Python 3 following this site

https://pypi.org/project/rpi-hardware-pwm/

It was working fine since September 2024. Initially I had to modify this boot/config.txt by adding dtoverlay=pwm-2chan. It worked fine, I created a virtual environment and everything went fine python programs were running smoothly generating PWMs.

Recently after 2 accidental power cuts to the Pi, when I ran the python program from the command line it gave an error message saying, add "dtoverlay=pwm-2chan".

Then I looked for boot/config.txt. When I opened this txt file it said it is read only and it cannot be edited. It also had words saying the file is moved to boot/firmware/config.txt.

But I checked this config.txt in boot/firmware location it already has "dtoverlay=pwm-2chan". I was so confused and thought something might have corrupted the system so I tried to load a back up OS to the SD card and found the same issue of config.txt moved from boot to boot/firmware. Then I did a complete re installing of OS on the SD card hoping that might solve the issue but the problem is the same.

Now I cant run any pwm programs because the initial step of adding adding dtoverlay=pwm-2chan to boot/config.txt itself cannot be performed because it is moved to another location and this one cannot be edited it is read only.

Please help me out is it possible to find a solution. Could there be a hardware problem after the power cuts since it was working before fine.

Thanks a lot.

electro201, 2025-04-05:
I found the problem. When latest updates are installed and the Pi is rebooted, then something messes with the GPIO pins or the way the config. txt works. I have not installed updates and ran it without latest updates everything works fine the way it used to.

Nothing wrong with config.txt or its location.

Hope this helps others too.

Thanks.

SaWaib, 2025-04-11:
I had the exact same Issue. For me it helped changing chip=2 to chip=0 as in "__init__.py the chippath is set to the following directory:"self.chippath: str = f"/sys/class/pwm/pwmchip{chip}"" .On my pi 5 only the directorys "/sys/class/pwm/pwmchip0" and "/sys/class/pwm/pwmchip1"  existed which is why setting "chip=2" resulted in an error.
Take this advise with caution as I am relatively unfamiliar with the library myself and not an expert in this topic

electro201, 2025-04-11:
` Take this advise with caution as I am relatively unfamiliar with the library myself and not an expert in this topic `

I am also not an expert so I do not wish to take ant risks. Thanks for sharing your knowledge and experience. 

I just wanted to inform the developer about this.

CamDavidsonPilon, 2025-05-23:
For modern RPiOS, RPi5 now uses chip 0

closed 2025-05-23 by CamDavidsonPilon